embeddedgyro/shakey-table #33: There needs to be a main program.

Adam-Englebright, 2024-03-30:
We need an actual program that utilises our libraries to interface with the MPU6050, INA260, and MD20, to drive our anti-quake cup holder. That is, read accel and gyro data from the MPU6050, calculate the cup holder angle/position, then drive the MD20 to produce the required corrective torque from the motor, using the INA260 data as torque feedback data.

genamag, 2024-04-13:
the main program is done and tested, everything works as intended after we  replaced faulty components

there is only a problem of occasional I2C errors, in sensor communication, but they do not disrupt the functioning of the main program and don't affect performance, as sensor readings are sent with very high freqeuncy

genamag, 2024-04-13:
Because we fried the INA260 current sensor, we can only control the wheel with PID control from gyro readings, so there is no good control over the torque produced by the motor

closed 2024-04-14 by genamag