embeddedgyro/shakey-table #49: Create a proper README file with hardware and software instruction

genamag, 2024-04-08:
Torin - CAD design
Gennady - Hardware setup
Adam - Software setup and installation
Jamby - Running Unit testing

genamag, 2024-04-13:
Added a hardware description with all components and what they are doing + how to connect. Also created and added images of the circuitry and positioning of key components

genamag, 2024-04-13:
yesterday we changed the position of the INA from before to after the motor to protect it from the reverse bias current created by the filter, so I need to reflect that in Wiki, change images and README in terms of this modification to the circuitry

genamag, 2024-04-13:
@dalvador2 How is the CAD and INA software part doing? 
@Jambu1234 How is Unit testing part doing? (don't forget about writing about CMake for them as well with instructions on how to run them?
@Adam-Englebright how is the MPU6050 and main program section doing?

Please, report on your progress

Adam-Englebright, 2024-04-13:
The section on setting up the software is essentially done. I want to add details about the lines required in `/boot/firmware/config.txt`.

Jambu1234, 2024-04-13:
Readme for Unit testing execution has been updated with the necessary cmake comments to run them.

dalvador2, 2024-04-14:
closed #61

closed 2024-04-14 by dalvador2